CANopenTerm/CANopenTerm #34: List of Scripts is empty

Sivet, 2024-07-01:
Hey, I've installed and run the CANopenTerm on a Ubuntu 24 following the install guide on https://canopenterm.de/getting-started.

It opens and gives a [SUCCESS] CAN successfully initialized
But when I press L to list the script an empty grid is shown.

[image]

I can see the scripts under /export/scripts but they seem to not be found by the program when running or maybe aren't added to the build?

And I should have a CAN network available.
[image]

mupfdev, 2024-07-01:
Hi Sivet,
thanks for your interest. On Linux there are currently several paths in which the [scripts](https://github.com/CANopenTerm/CANopenTerm/tree/main/export/scripts) are searched for:

```c
static const char* script_dirs[] = {
    "./",
    "./scripts",
    "/usr/share/CANopenTerm/scripts"
};
```

You should also copy the utility libraries from [export/lua](https://github.com/CANopenTerm/CANopenTerm/tree/main/export/lua) to `/usr/share/lua/5.4/lua`.

As I mainly use the Windows version myself, I don't yet have an elegant solution to this problem. If you have any ideas on how to make this work better, please let me know. Suggestions are very welcome.

Perhaps [CMake install](https://cmake.org/cmake/help/latest/command/install.html) would be an option?

Let me know if you need further assistance.

Best regards, Michael

mupfdev, 2024-07-01:
So if you cannot or do not want to install CANopenTerm system-wide, simply copy the binary to the `export` directory. CMake should do this automatically. If you then start it from this directory, it should just work.

Sivet, 2024-07-02:
Just copying the binary over seems to work perfectly. I'll look into installing it system-wide with CMake, but for now this is perfect.
Thanks for the quick help :)

closed 2024-07-02 by Sivet

mupfdev, 2024-07-02:
The CMake thing was more directed at myself. I have not yet defined any installation paths in CMake.

If you need further support or you find issues in the Linux version, please let me know. As I said, my main focus is on Windows, but I definitely want to continue to fully support Linux.

How and where did you come across CANopenTerm in the first place and, if I may ask, what do you want to use it for? Purely out of interest. :)

Best regards, Michael