CANopenTerm/CANopenTerm #84: Compilation error

corradosantoro, 2024-12-16:
There is the following problem when I try to compile it on Linux Ubuntu 22.04.1:

[ 48%] Building C object CMakeFiles/core.dir/src/api/lua_test_report.c.o
In file included from /home/corrado/Progetti_di_Ricerca/AgriMet/software/CANopenTerm/src/core/eds.h:14,
                 from /home/corrado/Progetti_di_Ricerca/AgriMet/software/CANopenTerm/src/api/lua_test_report.c:11:
/home/corrado/Progetti_di_Ricerca/AgriMet/software/CANopenTerm/src/core/codb.h:13:10: fatal error: cJSON.h: No such file or directory
   13 | #include "cJSON.h"
      |          ^~~~~~~~~
compilation terminated.

mupfdev, 2024-12-16:
I can replicate the problem and have a have a look. Thank you!

closed 2024-12-16 by mupfdev

mupfdev, 2024-12-18:
Did you get it to work?

commit 1e0313e6 (2024-12-16): Fix #84