CANopenTerm/CANopenTerm #92: Buildint CANopenTerm with llvm fails because of LTO usage in pocketpy

madebr, 2025-03-21:
When the toolchain supports it, pocketpy enables LTO eagerly.
Because CANopenTerm is  built with LTO disabled, linking CANopenTerm executables fails.

LTO in pocketpy can be disabled by configuring it with `-DPK_BUILD_WITH_IPO=FALSE`:
```patch
--- a/cmake/os_linux.cmake
+++ b/cmake/os_linux.cmake
@@ -153,6 +153,7 @@ ExternalProject_Add(pocketpy_devel
         -DPK_BUILD_STATIC_LIB=ON
         -DPK_ENABLE_OS=ON
         -DPK_BUILD_MODULE_LZ4=OFF
+        -DPK_BUILD_WITH_IPO=OFF
     INSTALL_COMMAND ${CMAKE_COMMAND} -E echo "Skipping install step."
     PATCH_COMMAND   ${CMAKE_COMMAND} -E echo "Skipping patch step.")
 
```

Also, llvm emits the following warning:
```
[92/100] Building C object CMakeFiles/core.dir/src/core/can_linux.c.o
/tmp/CANopenTerm/src/core/can_linux.c:147:1: warning: non-void function does not return a value in all control paths [-Wreturn-type]
  147 | }
      | ^
```
I think you should build this project with more warnings enabled (`-Wall -Wextra ...`) by default.

mupfdev, 2025-03-22:
Thanks!

PS: The warnings were already enabled:

https://github.com/CANopenTerm/CANopenTerm/blob/e17e0f16777e6fdef0ff8145ce3bed560fb1a2f0/CMakeLists.txt#L130-L135

madebr, 2025-03-22:
> PS: The warnings were already enabled:

These lines were not enabled on my system, since `-Werror` would have caused a failed build.

Also, enabling `-Werror` by default is not advises since it will suddenly fail when a new warning is introduced.
I'd suggest to enable it optionally, or by default. Or use the [`COMPILE_WARNING_AS_ERROR`](https://cmake.org/cmake/help/latest/prop_tgt/COMPILE_WARNING_AS_ERROR.html) property, which should support all toolchains.

commit 647bf492 (2025-03-22): Fix #92